iotappstory/ESP-Library #157: Get IAS root certificate not reacting

happertje2, 2020-05-21:
I am using the library 2.1.0 which i downloaded today of github.
By trying the sketch IASloader i want to try how the firmware upload works.

I have my device, app and project setup but i am not able to get a IAS root certificate which makes it impossible for the module to download the firmware. After exiting the configuration state, and having the module connected to the wifi it outputs over the serial monitor that it could not load certificate while it checks for updates of the sketch and SPIFFS.
I tried get IAS root certificate in the config state, i also tried uploading my own with use of the python script (CertToESP32) that Andreas Spiess made but no succes.

I also tried hardcoding the certificate as root_ca but that doesn't seem correct.
Could you help me in what i am doing wrong?

My module is a ESP-WROOM32 nodemcu but i think that shouldn't be the issue.

Onno-Dirkzwager, 2020-05-22:
Hi @happertje2,
when compiling your IASloader, which Partition Scheme did you select?

happertje2, 2020-05-22:
I tried it within a time of 8 hours. I tried it on chrome, internet explorer and android. I went for the partition minimal spiffs 1,9mb app with OTA and 190kb SPIFFS. And i chose this one because its the only one explicitly mentioning SPIFFS and OTA.

Onno-Dirkzwager, 2020-05-22:
@happertje2 both @CwlBroeders and me used the "Get IAS root cert." button a few times in the last 2-3 hours without problems. So we know it can work....Now we have to figure out why it doesn't work for you. 

Could you set the debug level to 3 in the config.h file. And copy the serial output when you press the "Get IAS root cert." button and when you callhome?

happertje2, 2020-05-22:
@Onno-Dirkzwager Updated the esp core from 1.0.2 to 1.0.4 and also tried it today, one of these 2 factors has made it work.

I have not started the debug, and i think it was indeed a network glitch which might have taken a bit too long. Thanks for the good care :D !!!

Onno-Dirkzwager, 2020-05-22:
@happertje2 
thanks for the feedback! 
Please close this issue (it's better for the stats if you do)

closed 2020-05-22 by happertje2