iotappstory/ESP-Library #158: Systematic lost of Certificate after OTA SPIFFS upload

FredericGuerne, 2020-06-06:
Hello,

Systematically I recieve the following message : 
"Could not load Certificate!" 
at next call home, for app. and SPIFFS OTA update, after a previous successful SPIFFS update.

I have to go back in config. mode to ask for a new certificate to make it working again
it looks like SPIFFS file overwrite the certificate data ?

My configuration : 
IotAppstory-ESP 2.1.0 RC3
ESP32
PlatformIO
Visual Studio Code

Best regards

Frederic

CwlBroeders, 2020-06-06:
Could u share how much memory u allocated to spiffs?

FredericGuerne, 2020-06-06:
My spiffs.bin is 1472 Kb, but I am not sure it is the allocated memory you talk about, sorry ?

CwlBroeders, 2020-06-06:
I was referring to the amount of flash memory set to be allocated to spiffs in your IDE settings, but i dont think thats the problem 

Have u added the certificate to the files that get compiled to spiffs.bin?

In the data folder u should make a folder named cert, this folder should contain the certificate, u can download it here.
https://iotappstory.com/ota/iasRootCa.cer

(its a recent change in workflow so we have not yet added it to the documentation properly)

CwlBroeders, 2020-06-06:
The IAS wiki has been updated to reflect changes.

https://iotappstory.com/wiki/arduino-esp/write-my-first-app/using-spiffs

SiskoBen, 2020-06-08:
Hi @CwlBroeders and @FredericGuerne ,

I can confirm that this instruction does in fact work. I also confirmed that omitting the /cert/iasRootCA.cer from the SPIFFS will in fact cause the certificate that was there when configuring the device to be lost and thus it won't successfully call home anymore until the certificate is added again from config mode. Adding the certificate in the SPIFFS does in fact fix the "lost certificate" problem.

Thanks for this question (I happen to have just started thinking about adding some SPIFFS functionlity to my sketch and @FredericGuerne 's question and @CwlBroeders answer came exactly at the right moment).

PS: using Platform IO with Atom as editor, IotAppstory-ESP 2.1.0 RC3 and and MH-ET Mini Live ESP32 in case that is relevant.

FredericGuerne, 2020-06-08:
Dear  CwlBroeders, 

thanks so much for your very good and quick support, it works now perfectly !!!

No lost of certificats after SPIFFS upload after multiple tests that's great !

I'm glad if my question helped other like SiskoBen, wonderful !

Be blessed

Frédéric

CwlBroeders, 2020-06-08:
@FredericGuerne Glad i could help, and thank you for the reminder to update the documentation :)

@SiskoBen Thank you for the verification

closed 2020-06-08 by CwlBroeders