iotappstory/ESP-Library #160: Platformio Support

mcsarge, 2020-06-20:
I am trying to use the library with some limited success. I have an ESP32 Development board and I am trying to figure out how to setup the SPIFFs in platformio - can you direct me to how to do that? I have looked in the docs in the platformio library and there is very little information. Usually you create a "data" directory and it is created for you, but I have not idea what goes in there or how to upload it to the website separately.

CwlBroeders, 2020-06-20:
Hello @mcsarge , the 'data' folder should contain any files u want to use in your app (ie. html/css/js for a web interface, or json/csv for data)
platform.io will produce a spiffs.bin file from the data folder and store it for later use.
I dont use platform.io myself but from [this question](https://community.platformio.org/t/how-to-make-bin-file-using-platformio-ide/8491) i would expect the spiffs.bin file should be stored in 
the .pioenvs / .pio/build / name-of-env folder of your project, once it has been produced.

mcsarge, 2020-06-20:
Thanks @CwlBroeders So, when I using ioappstory, what would I put in that data directory? In the instructions it talks about reserving space for files that are created by the library, I guess. Are there some default files I can load in the /data area that would allocate enough space?

CwlBroeders, 2020-06-20:
@mcsarge there is no actual need to put anything into the data directory, unless you have a need for it (like files for a web app or data to be used by your own code)

From the [IAS wiki](https://iotappstory.com/wiki/arduino-esp/write-my-first-app/using-spiffs)

"You don't need to compile or upload SPIFFS to make IOTAppStory work. But from library v2.1.0 on you do need to select at least 64Kb of SPIFFS for storing certificates in the IDE board settings. These certificates are used for TLS(https) communication with IOTAppStory.com"

The space for spiffs will be allocated in the ESP32 flash depending on your chosen [partition table](https://docs.platformio.org/en/latest/platforms/espressif32.html#partition-tables)

For an example using spiffs for storing files used in a webapp see the [WebAppToggleBtn](https://github.com/iotappstory/ESP-Library/tree/master/examples/WebAppToggleBtn) example.
(note the cert directory)

mcsarge, 2020-06-21:
OK! So in reading about partition tables in platformio, it specifies that when using the Arduino framework (I am) the default.csv is used and it defines a SPIFFS partition to be about 1.5 MB and that is certainly enough for IAS AppStory. Many thanks for the help!

mcsarge, 2020-06-21:
When using Platformio for the ESP32, the default partition table in the Arduino framework creates a SPIFFS partition that is large enough for IOTAppStory operation without any additional actions being taken.

closed 2020-06-21 by mcsarge