iotappstory/ESP-Library #165: Conflict boardnames

bergvdna, 2020-10-17:
Before update "Espressif ESP32 Dev Module" after update "ESP32 Dev Module".
No match anymore.

[image]
[image]
[image]

Please advise or correction.

Best regards.

CwlBroeders, 2020-10-17:
I am not sure what problem you are running into, could you elaborate?

bergvdna, 2020-10-18:
In Dutch. Hierna in Engels vertaald met Google translate.

Beste Christiaan,

Bedankt voor je snelle reactie.
Natuurlijk wil ik het uitgebreider beschrijven. Mijn Engels is niet zo
goed. Sorry.

Het probleem.
Ik gebruik Platformio voor mijn project. Daar kies ik mijn board.
De naam van het board in Platformio is Espressif ESP32 Dev Module.
Die naam van het board wordt door iotappstory kennelijk gebruikt voor een
match met iotappstory.com.
Zie de afbeelding (IOTAPPSTORY DEVICE REGISTRATION) die ik bij de issue heb
bijgesloten.
Er staat -Your device was defined as "Espressif ESP32 Dev Module" with 4 MB
flash-.
Dat is dus de naam van het board binnen Platformio.
Maar op de afbeelding staat onderaan: "We did not find any devboards or
modules that fit this description."
Tot voor kort werkte het nog fantastisch. Maar ik moest op iotappstory.com
wel een conversie uitvoeren
omdat er een wijziging is doorgevoerd in de structuur op de website /
database.
Als ik nu op de website kijk zie ik de naam van dit board ook niet meer
voorkomen.
Er staat nu als naam van het board "ESP32 Dev Module", dus ZONDER
-Espressif -.
Zie hiervoor de afbeelding (EDIT DEVICE).
Dat zal de reden zijn dat de match niet meer mogelijk is.

Dus hoe kan ik dat oplossen of moeten jullie de naam op iotapstory.com weer
herstellen?

Ik hoop dat het hiermee duidelijker is geworden.

Met vriendelijke groet,
Arie.

===================================
In Englisch.

Dear Christian,

Thanks for your quick response.
Of course I want to describe it in more detail. My English is not so good.
Sorry.

The problem.
I use Platformio for my project. There I choose my board.
The name of the board in Platformio is Espressif ESP32 Dev Module.
That name of the board is apparently used by iotappstory for a match with
iotappstory.com.
See the image (IOTAPPSTORY DEVICE REGISTRATION) that I have included with
the issue.
It says -Your device was defined as "Espressif ESP32 Dev Module" with 4 MB
flash-.
That is the name of the board within Platformio.
But the image at the bottom says, "We did not find any devboards or modules
that fit this description."
Until recently it worked fantastic. But I had to do a conversion on
iotappstory.com
because a change has been made to the structure on the website / database.
When I now look at the website I don't see the name of this board anymore.
It now says "ESP32 Dev Module" as the name of the board, so WITHOUT
-Espressif -.
See the image (EDIT DEVICE) for this.
That will be the reason that the match is no longer possible.

So how can I fix that or do you need to restore the name on iotapstory.com?

I hope it has become clearer with this.

Kind regards,
Arie.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virusvrij.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

Onno-Dirkzwager, 2020-10-19:
@bergvdna,
Thankyou for pointing this out. Ill answer in English for other users.... 
I did a little digging and think we both arrive at more or less the same conclusion.

It seems like PlatformIO includes a different  defined ARDUINO_BOARD for the board variants when compiling.
In your case we expected:
_Your device was defined as "ESP32_DEV" with 4 MB flash._
but got:
_Your device was defined as "Espressif ESP32 Dev Module" with 4 MB flash._

This was not a problem before. But we have become stricter since the last update in how we  link apps to devices based on this ARDUINO_BOARD define (which we store when you add your device)

Admittedly we did nearly all the final testing with Arduino IDE as this is where most of our users are. We should have been able to catch this..... and will update the testing procedures.

I expect to push an update solving this issue whithin 24-hours.

Onno-Dirkzwager, 2020-10-19:
@bergvdna @CwlBroeders,

Have you modified anything in you PIO settings regarding your ESP board?

Because accourding to [this link](https://community.platformio.org/t/wrong-board-name-for-arduino-boards/13395/2) at PIO your board should have been defined as:
PLATFORMIO_     followed by the normal Arduino board def.

Resulting in
_Your device was defined as "PLATFORMIO_ESP32_DEV" with 4 MB flash._
and not:
_Your device was defined as "Espressif ESP32 Dev Module" with 4 MB flash._

You still found a bug: In our previous versions of the website we auto removed the "PLATFORMIO_" prefix from the ARDUINO_BOARD define. And in the new one this is missing.

But this is not going to solve your issue..... We have to figure out why the apps you compile with PIO end up with the wrong define.

bergvdna, 2020-10-20:
> Have you modified anything in your PIO settings regarding your ESP board?

Dear Onno, Christiaan.

Q: Have you modified anything in your PIO settings regarding your ESP board?
A: No.

I see two options in the e-mails.
1.  19 okt. 2020 21:14   : Because accourding to this link
<https://community.platformio.org/t/wrong-board-name-for-arduino-boards/13395/2>
at
PIO your board should have been defined as ........ :But this is not going
to solve your issue..... We have to figure out why the apps you compile
with PIO end up with the wrong define.

2.  19 okt. 2020 20:32  : Admittedly we did nearly all the final testing
with Arduino IDE as this is where most of our users are. We should have
been able to catch this..... and will update the testing procedures.

I expect to push an update solving this issue whithin 24-hours.

Is the answer in 19 okt. 2020 20:32 the right one?
Can i start testing tomorrow?

Best regards,

Arie.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virusvrij.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

Onno-Dirkzwager, 2020-10-20:
@bergvdna 

We pushed our fix re-implementing the "remove PLATFORMIO_ prefix". So please try. But as mentioned before I don't expect this to fix your problem as you have an entirely wrong ARDUINO_BOARD define.

If all is setup correctly you should now get:
Your device was defined as "PLATFORMIO_ESP32_DEV" with 4 MB flash.
and not:
Your device was defined as "Espressif ESP32 Dev Module" with 4 MB flash.

Our fix transforms PLATFORMIO_ESP32_DEV to the expected: ESP32_DEV
But then again your ARDUINO_BOARD define: "Espressif ESP32 Dev Module" worries me ;)

Maby @Valdhor can help by comparing PIO board setup?

It's been a while since I used PlatformIO... But I decided to revisit this topic and write a tutorial this weekend on how to setup plaformio for esp's using IAS.

So Ill let you know if I find anything interesting.

bergvdna, 2020-10-21:
The original text in Dutch is below.

Dear Onno,

I tested it this morning at about 8:30 am.

Unfortunately with a negative result and the same message: No match.

I have added the images again.

Also an image of my platformio.ini and also the board choice when creating
a new project.

As you can see, the last call home took place on 12-10-2020.
This does NOT mean that it stopped working on 13-10-2020 by the way.
Haven't used it for a while until this week.

I hope you know enough about this.

If I can look up something about the board in Platformio (I don't know
where), feel free to give some pointers.
Then you will still receive that information.

Now included 5 images of which 1 Platformio (platformio.ini) and 1
Platformio (choose the same board for new project).

Kind regards,
Arie van den Berg.

==================================

Beste Onno,

Ik heb het vanmorgen om ongeveer 8.30 uur getest.

Helaas met een negatief resultaat en de zefde melding: Geen match.

Ik heb de afbeeldingen er weer bij gedaan.

Ook een afbeelding van mijn Platformio.ini en ook de board keuze bij de
aanmaak van een nieuw project.

Zoals je kunt zien heeft op 12-10-2020 de laatste callhome nog
plaatsgevonden.
Dit betekent NIET dat het op 13-10-2020 niet meer werkte trouwens. Ik heb
het een tijdje niet gebruikt tot deze week.

Ik hoop dat je hier voldoende door weet.

Als ik nog iets op kan zoeken over het board in Platformio (ik weet niet
waar) geef gerust nog wat aanwijzingen.
Dan krijg je die informatie alsnog.

Nu bijgesloten 5 afbeeldingen waarvan 1 Platformio (platformio.ini) en 1
Platformio (keuze zelfde board bij nieuw project).

Met vriendelijke groet,
Arie van den Berg.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virusvrij.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

Valdhor, 2020-10-21:
My current board is the Heltec WiFi Kit 32 and here is my platformio.ini:

`[env:heltec_wifi_kit_32]
platform = espressif32
board = heltec_wifi_kit_32
framework = arduino
monitor_speed = 115200
lib_deps = 
	me-no-dev/AsyncTCP@^1.1.1
	me-no-dev/ESP Async WebServer@^1.2.3
	iotappstory/IOTAppStory-ESP@^2.1.0-RC3`

I did have to make sure my lib_deps were in the correct order before it would compile. I had made the assumption that it didn't matter but it appears that it does.

I'm afraid I don't see yours.

Valdhor, 2020-10-21:
That does not look right:

[env:heltec_wifi_kit_32]
platform = espressif32
board = heltec_wifi_kit_32
framework = arduino
monitor_speed = 115200
lib_deps = 
	me-no-dev/AsyncTCP@^1.1.1
	me-no-dev/ESP Async WebServer@^1.2.3
	iotappstory/IOTAppStory-ESP@^2.1.0-RC3

Onno-Dirkzwager, 2020-12-04:
@bergvdna I owe you an apology!
- for taking so long
- you were right
Ive been working with some people on Discord to figure this out. And as it turns out PIO is very inconsistent when compiling the ARDUINO_BOARD define. With some boards it uses the board name, with others the variant or other fields.

So in the end you where right! We need the user to be able to select boards / modules if undetected because chances are it is a supported board / module.

This has been changed, feel free to reopen the issue if deemed necessary.

closed 2020-12-04 by Onno-Dirkzwager