iotappstory/ESP-Library #167: Doesn’t work with PlatformIO

Valdhor, 2020-10-17:
Should this work with PlatformIO? When I try to compile the initial application I get “Can’t find FS.h”. I don’t know where to find that library.

CwlBroeders, 2020-10-17:
This is the ESP arduino core file system library (it replaces the SPIFFS library).
https://github.com/esp8266/Arduino/blob/master/cores/esp8266/FS.h

Valdhor, 2020-10-18:
It looks like this is for esp8266 but I have esp32. Is this still correct?

CwlBroeders, 2020-10-18:
For ESP32 this should also be present in the core framework.
https://github.com/espressif/arduino-esp32/blob/master/libraries/FS/src/FS.h

https://community.platformio.org/t/errors-when-compiling-esp32-code/11421

Valdhor, 2020-10-18:
Thanks for that. I will give it a try.

I’m thinking the question is why I need to install that as a separate install? Perhaps an update to the documentation?

Valdhor, 2020-10-18:
I do appreciate the quick response.

CwlBroeders, 2020-10-18:
The documentation is (for now) mainly tailored to the arduino IDE, aswell as being slightly outdated.

Valdhor, 2020-10-19:
Figured out you have to have lib_deps in the right order.

closed 2020-10-19 by Valdhor