iotappstory/ESP-Library #168: Reboot while in Config mode

NeliusNDL, 2020-11-12:
@Onno-Dirkzwager 

Hello Onno, I get this reboot sequence on some of my ESP32, seems something maybe interfere with OLD SPIFS ?
I then load the example Blink from Arduino on this board and I still get the same problem.

Problem: When Config mode entered, as soon as I try to use my smartphone (Sam Note 9) to connect to the WiFi AP "iasblink-912" the ESP32 reboots and the terminal shows the below log.

I am using ESP32-Wroom-32D

11:39:42.293 ->  C O N F I G U R A T I O N   M O D E
11:39:42.387 ->  AP mode. Connect to Wifi AP "iasblink-912"
11:39:42.387 ->  And open 192.168.4.1
11:39:46.559 -> dhcps: send_offer>>udp_sendto result 0
11:39:46.605 -> Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.
11:39:46.605 -> Core 0 register dump:
11:39:46.605 -> PC      : 0x00000000  PS      : 0x00060330  A0      : 0x80148f79  A1      : 0x3ffb54c0  
11:39:46.652 -> A2      : 0x3ffd32c0  A3      : 0x3ffd34f4  A4      : 0x3ffd393c  A5      : 0x3ffd391c  
11:39:46.652 -> A6      : 0x0204a8c0  A7      : 0x0c04a8c0  A8      : 0x80148e1c  A9      : 0x3ffb5480  
11:39:46.652 -> A10     : 0x3ffd32d0  A11     : 0x3ffd34f4  A12     : 0x3ffb54cc  A13     : 0x00000044  
11:39:46.652 -> A14     : 0x00000001  A15     : 0x00000006  SAR     : 0x00000010  EXCCAUSE: 0x00000014  
11:39:46.652 -> EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000  
11:39:46.698 -> 
11:39:46.698 -> Backtrace: 0x00000000:0x3ffb54c0 0x40148f76:0x3ffb5500 0x40155eb5:0x3ffb5520 0x4015aebd:0x3ffb5560 0x4016015a:0x3ffb5580 0x401496e3:0x3ffb55a0 0x40088b9d:0x3ffb55d0
11:39:46.698 -> 
11:39:46.698 -> Rebooting...

NeliusNDL, 2020-11-12:
I think In above example I put the wrong WiFi password. If the ESP then reboots into config mode that happens.

Now I have tried to clear the EEPROM to try with the fallowing command:
IAS.preSetWifi("BackupWIFI","BackupPW");  
When IAS.preSetWifi is called the CPU reboots with the following error. Looks like EEPROM issue, in this case it is CPU 1 that crashes. When rebooting in Config mode it is CPU 0 that crashes.

WIFICONNECTOR_DEBUG	| running getAPfromEEPROM(1)
WIFICONNECTOR_DEBUG	| Searching for magicBytes(CFG) at EEPROM address 115
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40001277  PS      : 0x00060030  A0      : 0x800d644d  A1      : 0x3ffd2f90  
A2      : 0x00000000  A3      : 0x3f401546  A4      : 0x1022e78a  A5      : 0x00000001  
A6      : 0x00060e20  A7      : 0x00000000  A8      : 0x1022e78a  A9      : 0x3ffd2ee0  
A10     : 0x3ffc1600  A11     : 0x3ffd2f05  A12     : 0x00000000  A13     : 0x00000047  
A14     : 0x00000073  A15     : 0x01000000  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000  

Backtrace: 0x40001277:0x3ffd2f90 0x400d644a:0x3ffd2fa0 0x400d28c9:0x3ffd2fe0 0x40088ba1:0x3ffd3000

Rebooting...

Onno-Dirkzwager, 2020-12-16:
@NeliusNDL should be fixed with the last pr, reopen this issue if necessary

closed 2020-12-16 by Onno-Dirkzwager