iotappstory/ESP-Library #174: Problem with connection to internet to Adafruit IO server

mamech, 2021-01-07:
Hi there

First of all I should show my appreciation of the nice idea and nice service of IoTAppStory, really helpful.

I tried to integrate a code I develop for IoT application using Adafruit IO server with your IoTAppStory. I noticed the following problem.
I connect to AdafruitIO using:```
```

  AdafruitIO_WiFi io(IO_USERNAME, IO_KEY, WIFI_SSID, WIFI_PASS);
  io.connect();`
  // wait for a connection
  while (io.status() < AIO_CONNECTED)
  {
    mySerial.print(".");
    delay(500);
  }
```

and it connects smoothly, but if I call IAS.begin();  before or even after the above shown code, the connection to adafruit IO never happens and it keep trying forever.

I think that this may be due some internal declarations from IAS.begin() . What looks weird to me, is that that IAS.begin() is blocking connection to server wherever it is called, even if it is called after code that is used for connection. Dos this mean that something happens during compilation that corrupts any sort of internet connection other than the one used with IoTAppStory?

CwlBroeders, 2021-01-08:
There sounds like an issieu with both IAS and adafruit IO wanting to make use of the WIFI at the same time.

Have you tried using "IAS.WiFiDisconnect();" to free the connection for  adafruits " io.connect;" ?

mamech, 2021-01-08:
> 
> 
> There sounds like an issieu with both IAS and adafruit IO wanting to make use of the WIFI at the same time.
> 
> Have you tried using "IAS.WiFiDisconnect();" to free the connection for adafruits " io.connect;" ?

Sure, using  "IAS.WiFiDisconnect();"  is the first thing that came to my mind, and it did not work too. As I noticed later, the origin of problem seems unclear to me, because if the problem is  IAS and adafruit IO wanting to make use of the WIFI at the same time, then the problem will appear after I call IAS, but what happens, is that even if IAS.begin() is called in code after Adafruit IO code related to connection, the connection fails.

CwlBroeders, 2021-01-08:
Could you put up a link to the adafruit library you are using?
And what version of the IAS library are you using?

mamech, 2021-01-08:
yes sure
IAS library 2.1.0-RC4 
and 
Adafruit IO Arduino library 3.9.1

CwlBroeders, 2021-01-08:
Looking at the code of the adafruit library i see it uses MQTT to communicate with the adafruit server.
https://github.com/adafruit/Adafruit_IO_Arduino/blob/master/src/wifi/AdafruitIO_ESP32.h#L21

(forgot to ask, but im presuming an ESP32 here)

Do not use the " setCallHomeInterval();" as this will cause problems with the MQTT connection to the adafruit.io service.

Instead, use IAS.CallHome(); in your code at the apropriate time (whenever you decide to update).
This you the oppertunity to turn off adafruit.io, make the callhome, and then trun adafruit.io on again.

Also use a delay after IAS.begin so the code has some time to finish running.

mamech, 2021-01-08:
I am using ESP8266. 
First of all I commented all of the code related to IAS, and just left these 2 lines "After" adafruit io call :
`
  IOTAppStory IAS(COMPDATE, MODEBUTTON);  // Initialize IotAppStory

  IAS.begin();                          
 // IAS.setCallHomeInterval(0);           
 // IAS.eepFreeFrom;
 // Serial.print(IAS.eepFreeFrom);
  //IAS.WiFiDisconnect();
`

Even if Those 2 lines are After calling adafruit io, the connection of adafruit does not proceed.This means, that the code is being stuck in while loop before even it reaches IAS.begin():
`
  while (io.status() < AIO_CONNECTED) // This while loop keeps looping forever if IAS.begin() is called, no difference IAS.begin is called before or after the loop
  {
    mySerial.print(".");
    delay(500);
  }

`
When I commented IAS.begin(). the adafruit while loop terminates normally and adafruit gets connected and other parts of code are executed.

In previous trials, I started my code with IAS.begin() and other necessary functions. It worked nicely and I could upload codes over internet, but when adafruit connection code comes, it never connects to server. so what appears to me, that IAS.begin() can corrupt adafruit connection regardless where IAS.begin() was called, and this is very strange point. Also as I could see, adafruit call does not corrupt IAS.begin().
Are not there any way to start with IAS.begin() and other necessary functions of IAS, then destroy everything related to wifi connection that were used during IAS, so I can start adafruit on clean ground?

CwlBroeders, 2021-01-08:
We have had several connection issieus resembling this behavior with ppl using MQTT connections in the past, usually the awnser to this was seperating the use of the wifi connections.

Could you add delay(2000); after IAS.begin();
(2 seconds delay is arbitrary number, but it will ensure the code is finished running).

CwlBroeders, 2021-01-08:
If this doesnt work, would you mind posting a barebones version of your code so we can have a look at it here?

Onno-Dirkzwager, 2021-01-19:
Hi @mamech any progress with this issue?

mamech, 2021-01-19:
Sorry for long silence. I was just busy with some things. I am preparing backbone version,  but just I noticed that its performance little bit different from the problem happening with original code. Anyway I will make some more tests and give you the code with my remarks.

mamech, 2021-01-26:
ok after several trials, the backbone version has somewhat different performance from the full code I was testing before, but from what I see now, Adafruit io connects well weather before or after the IAS call, the problem only now happens with IAS itself if it is called after adafruit connection.

`// Adafruit IO Publish Example
//
// Adafruit invests time and resources providing this open source code.
// Please support Adafruit and open source hardware by purchasing
// products from Adafruit!
//
// Written by Todd Treece for Adafruit Industries
// Copyright (c) 2016 Adafruit Industries
// Licensed under the MIT license.
//
// All text above must be included in any redistribution.

/************************** Configuration ***********************************/

// edit the config.h tab and enter your Adafruit IO credentials
// and any additional configuration needed for WiFi, cellular,
// or ethernet clients.
#include "config.h"  // this .h file has login credentials of wifi and adafruit io account
#define COMPDATE __DATE__ __TIME__
#define MODEBUTTON 0                    // Button pin on the esp for selecting modes. D3 for the Wemos!

#include <IOTAppStory.h>                // IotAppStory.com library

/************************ Example Starts Here *******************************/

// this int will hold the current count for our sketch
int count = 0;

// set up the 'counter' feed
AdafruitIO_Feed *counter = io.feed("compressor-1");

void setup() {

  IOTAppStory IAS(COMPDATE, MODEBUTTON);  // Initialize IotAppStory

  IAS.begin();                          // Run IOTAppStory
  // start the serial connection
  Serial.begin(115200);

  // wait for serial monitor to open
  while (! Serial);

  Serial.println("Connecting to Adafruit IO");

  // connect to io.adafruit.com
  io.connect();

  // wait for a connection
  while (io.status() < AIO_CONNECTED) {
    Serial.print(".");
    delay(500);
  }

  // we are connected
  Serial.println();
  Serial.println(io.statusText());

}

void loop() {

  // io.run(); is required for all sketches.
  // it should always be present at the top of your loop
  // function. it keeps the client connected to
  // io.adafruit.com, and processes any incoming data.
  io.run();

  // save count to the 'counter' feed on Adafruit IO
  Serial.print("sending -> ");
  Serial.println(count);
  counter->save(count);

  // increment the count by 1
  count++;

  // Adafruit IO is rate limited for publishing, so a delay is required in
  // between feed->save events. In this example, we will wait three seconds
  // (1000 milliseconds == 1 second) during each loop.
  delay(3000);

}`

mamech, 2021-01-26:
by the way, I updated Adafruit io library to be 4.0.2

CwlBroeders, 2021-01-27:
@mamech 
Thank you for the code.
I dont have much time at the moment but i will look into it after the weekend to see what is going on.