iotappstory/ESP-Library #175: compiling IASLoader

donbrew1, 2021-01-22:
I have tried many versions of the library and ESP8266 and Wemos D1 mini and several of the examples.
ALL of the examples produce an error failure to compile for board.

CwlBroeders, 2021-01-22:
Could you post the complete error here?
It should give us some clues as to what is wrong, possibly some depandancy missing.

donbrew1, 2021-01-22:
It is Exit Status 1 "Error compiling for board" then defines the particular board I am trying to load.  NodeMCU 1.0(ESP 12E module) or LOLIN(WEMOS) DI R@ & mini.

Other sketches load fine.

CwlBroeders, 2021-01-22:
Are you in our discord aswell?
If so, could you post a screenshot of the error output there?

donbrew1, 2021-01-22:
Arduino: 1.8.13 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Disabled, All SSL ciphers (most compatible), 4M (no SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

C:\Program Files\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\Arduino\hardware -hardware C:\Users\donwin7\AppData\Local\Arduino15\packages -tools C:\Program Files\Arduino\tools-builder -tools C:\Program Files\Arduino\hardware\tools\avr -tools C:\Users\donwin7\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files\Arduino\libraries -libraries C:\Users\donwin7\Documents\Arduino\libraries -fqbn=esp8266:esp8266:d1_mini:xtal=80,vt=flash,exception=disabled,ssl=all,eesz=4M,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=921600 -vid-pid=1A86_7523 -ide-version=10813 -build-path C:\Users\donwin7\AppData\Local\Temp\arduino_build_705043 -warnings=none -build-cache C:\Users\donwin7\AppData\Local\Temp\arduino_cache_198073 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\donwin7\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-3-20ed2b9.path=C:\Users\donwin7\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9 -prefs=runtime.tools.mkspiffs.path=C:\Users\donwin7\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-3-20ed2b9 -prefs=runtime.tools.mkspiffs-2.5.0-3-20ed2b9.path=C:\Users\donwin7\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-3-20ed2b9 -prefs=runtime.tools.python.path=C:\Users\donwin7\AppData\Local\Arduino15\packages\esp8266\tools\python\3.7.2-post1 -prefs=runtime.tools.python-3.7.2-post1.path=C:\Users\donwin7\AppData\Local\Arduino15\packages\esp8266\tools\python\3.7.2-post1 -verbose C:\Users\donwin7\Documents\Arduino\libraries\IOTAppStory-ESP\examples\IASLoader\IASLoader.ino

C:\Program Files\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files\Arduino\hardware -hardware C:\Users\donwin7\AppData\Local\Arduino15\packages -tools C:\Program Files\Arduino\tools-builder -tools C:\Program Files\Arduino\hardware\tools\avr -tools C:\Users\donwin7\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files\Arduino\libraries -libraries C:\Users\donwin7\Documents\Arduino\libraries -fqbn=esp8266:esp8266:d1_mini:xtal=80,vt=flash,exception=disabled,ssl=all,eesz=4M,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=921600 -vid-pid=1A86_7523 -ide-version=10813 -build-path C:\Users\donwin7\AppData\Local\Temp\arduino_build_705043 -warnings=none -build-cache C:\Users\donwin7\AppData\Local\Temp\arduino_cache_198073 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\donwin7\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-3-20ed2b9.path=C:\Users\donwin7\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9 -prefs=runtime.tools.mkspiffs.path=C:\Users\donwin7\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-3-20ed2b9 -prefs=runtime.tools.mkspiffs-2.5.0-3-20ed2b9.path=C:\Users\donwin7\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-3-20ed2b9 -prefs=runtime.tools.python.path=C:\Users\donwin7\AppData\Local\Arduino15\packages\esp8266\tools\python\3.7.2-post1 -prefs=runtime.tools.python-3.7.2-post1.path=C:\Users\donwin7\AppData\Local\Arduino15\packages\esp8266\tools\python\3.7.2-post1 -verbose C:\Users\donwin7\Documents\Arduino\libraries\IOTAppStory-ESP\examples\IASLoader\IASLoader.ino

Using board 'd1_mini' from platform in folder: C:\Users\donwin7\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2

Using core 'esp8266' from platform in folder: C:\Users\donwin7\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2

Detecting libraries used...

"C:\\Users\\donwin7\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Users\\donwin7\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.2/tools/sdk/include" "-IC:\\Users\\donwin7\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.2/tools/sdk/lwip2/include" "-IC:\\Users\\donwin7\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.2/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\donwin7\\AppData\\Local\\Temp\\arduino_build_705043/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=c++11 -ffunction-sections -fdata-sections -fno-exceptions -w -x c++ -E -CC -DNONOSDK221=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10813 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_WEMOS_D1MINI\"" -DFLASHMODE_DIO -DESP8266 "-IC:\\Users\\donwin7\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.2\\cores\\esp8266" "-IC:\\Users\\donwin7\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.2\\variants\\d1_mini" "C:\\Users\\donwin7\\AppData\\Local\\Temp\\arduino_build_705043\\sketch\\IASLoader.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE

Alternatives for IOTAppStory.h: [IOTAppStory-ESP@2.1.0-RC4]

ResolveLibrary(IOTAppStory.h)

  -> candidates: [IOTAppStory-ESP@2.1.0-RC4]

[... cut ...]
Using library IOTAppStory-ESP at version 2.1.0-RC4 in folder: C:\Users\donwin7\Documents\Arduino\libraries\IOTAppStory-ESP 

Using library ESP8266WiFi at version 1.0 in folder: C:\Users\donwin7\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi 

Using library EEPROM at version 1.0 in folder: C:\Users\donwin7\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\EEPROM 

Using library ESP8266mDNS at version 1.2 in folder: C:\Users\donwin7\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266mDNS 

exit status 1

CwlBroeders, 2021-01-22:
C:\Users\donwin7\Documents\Arduino\libraries\IOTAppStory-ESP\src/espressif/ConfigServer.h:27:73: fatal error: ESPAsyncTCP.h: No such file or directory

There is a dependancy missing that the IAS library needs 
https://iotappstory.com/wiki/arduino-esp/how-to-get-started/prerequisites

if you install this, it should compile as expected
https://github.com/me-no-dev/ESPAsyncTCP//

donbrew1, 2021-01-22:
Installed and compiling now.

Thank you

CwlBroeders, 2021-01-22:
You are welcome, good luck with your projects  :)

closed 2021-01-22 by CwlBroeders