iotappstory/ESP-Library #176: IASConfig page in Google chrome (Android 8) is rendered with errors

AlexZibin, 2021-02-20:
Please take a look at enclosed image - page icons overlap over the title text:
[image]

CwlBroeders, 2021-02-20:
Noted, thanks for letting us know

Onno-Dirkzwager, 2021-02-21:
@AlexZibin we will fix the cloud version of these files this week. SPIFFS verion in the next release.

Onno-Dirkzwager, 2021-02-25:
Hi @AlexZibin 
I've been looking into this but unfortunately I cannot replicate this issue.
Android 8 - Chrome
[image]

When using devtools to analyse resizing and icon placement all seems to scale well as show below (bad capture but gives an idea.

[image]

So untill I know more and can replicate it. There's not much I can do.

Onno-Dirkzwager, 2021-02-25:
Does anybody else have this issue?
[image]

Onno-Dirkzwager, 2021-03-05:
@AlexZibin did you have time to read my comments?

AlexZibin, 2021-03-05:
> @AlexZibin did you have time to read my comments?
Hi Onno, 
Thank you for your support. 
Yes, I did. 
I can test updated lib tomorrow, and will reply ASAP.

Onno-Dirkzwager, 2021-03-06:
Hi @AlexZibin thankyou for your reply. 
I did not change anything as I could NOT replicate this. As explained above I have tested on multiple devices including an Android 8.0.0 with chrome.

That's not to say that is not a real issue. But I need to be able to replicate it, to study it and come up with a solution. Any extra feedback is appriciated.

AlexZibin, 2021-03-06:
Hi Onno, 
It you will need my help please don't hesitate to write me.

Onno-Dirkzwager, 2021-03-06:
@AlexZibin 
maybe you can help me track down the differences between our phones. 
We both have Android 8.0.0 with chrome
- which chrome version do you have?
- have you customized your chrome display settings? Font-size etc.
- could you send me an unedited screenshot so I can figure out your display size?

Which IAS library version are you using?

Did you have the same issue on other devices?

Any chance you could do the same test on another phone to compair?

AlexZibin, 2021-03-07:
Hi Onno,
> 
> We both have Android 8.0.0 with chrome

I have Android 8.1.0 on ZTE Blade V0920

> * which chrome version do you have?

Chrome 88.0.4324.181

> * have you customized your chrome display settings? Font-size etc.

No

> * could you send me an unedited screenshot so I can figure out your display size?

Please find below, sized 720*1440
[image]
> 
> Which IAS library version are you using?

<IOTAppStory-ESP> 2.1.0-RC4
> Did you have the same issue on other devices?
> Any chance you could do the same test on another phone to compair?

I can do more tests after March 15th. From tomorrow till 15th I will go for vacations and will not have enough time to response.
I've tried today but I get this error when trying to sign in:
[image]