iotappstory/ESP-Library #177: Incompatible with github.com/s00500/ESPUI?

AlexZibin, 2021-02-20:
First of all, great THANK YOU for this useful library.

Please help to combine (in one sketch) two must-have libraries:
  https://github.com/iotappstory/ESP-Library
    and
  https://github.com/s00500/ESPUI  (it's a dashboard like `ayushsharma82/ESP-DASH` with rich interface)

My trial sketch is hosted at https://github.com/AlexZibin/RoboKite-Base-4--IAS-ESPUI
Sketch crashes when I press the Update button (which calls IAS.callHome () in line 58 of espui1.cpp) **with the following dump:**

```
Button UP
 Calling Home

 Checking for App(Sketch) updates from: https://iotappstory.com/ota/updates
 No updates necessary!

 Checking for SPIFFS updates from: https://iotappstory.com/ota/updates
E (62502) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (62502) task_wdt:  - async_tcp (CPU 0/1)
E (62502) task_wdt: Tasks currently running:
E (62502) task_wdt: CPU 0: IDLE0
E (62502) task_wdt: CPU 1: async_tcp
E (62502) task_wdt: Aborting.
abort() was called at PC 0x400ded4b on core 0

Backtrace: 0x4008c454:0x3ffbe170 0x4008c685:0x3ffbe190 0x400ded4b:0x3ffbe1b0 0x40084771:0x3ffbe1d0 0x401a69cb:0x3ffbc450 0x400e0102:0x3ffbc470 0x4008a381:0x3ffbc490 0x40088b9d:0x3ffbc4b0

Rebooting...
```

I understand that `AsyncWebServer` and `AsyncWebSocket` from `ESPUI` must be paused for the time when `IAS.callHome ()` works, but I haven't got enough experience to do it by myself.

Please help me to find the proper solution!

Onno-Dirkzwager, 2021-02-21:
@AlexZibin you have 2 options:

**Make you own function**
- That disables whatever is nessecairy
- runs IAS.callhome();
- Enables whatever you disabled after the callhome

**Use callbackes**
- Study Virginsoil full for callbacks. https://github.com/iotappstory/ESP-Library/blob/master/examples/VirginSoil-Full/VirginSoil-Full.ino#L93
- Disables whatever is nessecairy in the IAS.onFirmwareUpdateCheck  callback
- Enable whatever you disabled in the IAS.onFirmwareNoUpdate, IAS.onFirmwareUpdateSuccess & IAS.onFirmwareUpdateError callbacks

I would go for the first option.

AlexZibin, 2021-03-05:
Yes, thank you, I will try to fix.