iotappstory/ESP-Library #182: pwm not working at GPIO14 (ESP23-WROOM-32D)

SteffenBis, 2021-03-12:
I am not sure if this is really an issue with iotappstory, however I have some problems getting the pwm functionality to work on GPIO14. While trying to reproduce the issue with a smaller sketch I noticed that the problem does not occur at GPIO02 (built in LED of FireBeetle-Board V4.0 that I use).
Using the attached sketch the problem occurs only at GPIO14 and with "INCLUDE_OTA_UPDATE" set to "true". There is still a timer interrupt inside the sketch - for some reason I could not get rid of it without stopping the code from working (I dont know why, but maybe this is part of the issue?). In this example I used the jled-library for convenience (as I do in the original sketch). However it does not seem that the issue is linked to the jled-library. I do not use the default partition, but "min_spiffs.csv", since I ran out of flash memory with my bigger sketch.

[main.txt](https://github.com/iotappstory/ESP-Library/files/6131157/main.txt)
[platformio.txt](https://github.com/iotappstory/ESP-Library/files/6131168/platformio.txt)

Onno-Dirkzwager, 2021-03-12:
@SteffenBis thank you for this information Ill have a look.

SteffenBis, 2021-03-12:
I modified the code and got rid of the jled-library and using ledcWrite instead - this works now. So I guess it is some compatibility issue between jled and iotappstory that somehow does not affect GPIO02 but GPIO14.
Modified code:

#define TEST_LED 14                  
#define INCLUDE_OTA_UPDATE true      

#define COMPDATE __DATE__ __TIME__
#define MODEBUTTON 5

#include <Arduino.h>

#if INCLUDE_OTA_UPDATE
  #include <IOTAppStory.h>
  IOTAppStory IAS(COMPDATE, MODEBUTTON);
#endif

hw_timer_t * timer = NULL;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;

void IRAM_ATTR onTime() {
	portENTER_CRITICAL_ISR(&timerMux);
	// set some flags
	portEXIT_CRITICAL_ISR(&timerMux);
}

byte dutycycle;

void setup() {
  // for some reason the code does not run without this timer attatched (dont really understand - maybe linked to the issue?)
  timer = timerBegin(0, 80, true);                               
  timerAttachInterrupt(timer, &onTime, true);
  Serial.begin(115200);
  Serial.println("Start");
  ledcSetup(0, 5000, 8);
  ledcAttachPin(TEST_LED, 0);
}

void loop() {
  ledcWrite(0, dutycycle);
  dutycycle++;
  delay(5);
}

Onno-Dirkzwager, 2021-03-13:
@SteffenBis thank you for putting in the extra effort to find out where it goes wrong. I'll still check if there is anything we can do..... and leave this issue open as a reminder.

1JonaB1, 2021-03-15:
@SteffenBis  As far as i know pin 14 and some others do not work for PWM if Wifi is enabled.

SteffenBis, 2021-03-15:
@1JonaB1 interesting, thank you for pointing me to that thread. Over there somebody describes that ADC2 (the one for pin 14) is also unavailable if BLE is active. In my case PWM on pin 14 worked fine whith BLE running until I included the WiFi functionality. But as it seems it may be just a bad idea to use pin 14 with a fancy breathing LED while WiFi is active.

Onno-Dirkzwager, 2021-03-17:
Thankyou @1JonaB1 for sharing this information.
@SteffenBis I think we can close this issue now?

closed 2021-03-17 by SteffenBis