iotappstory/ESP-Library #186: Defining two AddField values with Interval field types results in second instance reverting to 1 minute

larz99, 2021-04-03:
Two Interval type settings fields does not appear to function properly.

Example:
  IAS.addField(updateInterval, "Update Interval", 7, 'I');
  IAS.addField(callHomeInterval, "Call Home Interval", 7, 'I');

When in Config mode, web page will accept values for both Interval fields, but switching to another tab then returning to the App Settings tab shows the second value reverting to 1 minute.

Onno-Dirkzwager, 2021-04-04:
@larz99 thank you for reporting!
As the config mode files get loaded from the cloud, this will be reasonably easy to fix, without having to wait for the next library release. I expect to fix this somewhere next week.

Onno-Dirkzwager, 2021-04-04:
@larz99 had some time available earlier as expected and fixed this bug.

Please confirm and close this issue if solved. Browsers have a tendency to cache js files. So make sure to refresh a few times or even empty your cache before testing.

larz99, 2021-04-07:
@Onno thank you for fixing!

This looks good.  I tested two interval fields, both were set independently.  I also exited and re-entered config mode, values were retained and correct.

I consider it fixed.

Many Thanks!

larz99, 2021-04-13:
Hi Onno,

I have worked with more interval type fields and find that my earlier testing was not complete.  Interval fields are still are not behaving properly.

1. With two interval fields defined, set the first to 2 minutes and the second to 1 days.
2. Press Save and note that values appear correct on the Settings tab.
[image]
3. Switch to another tab and then return to the Settings tab.
4. Note that the first interval field now diaplays 1 Minutes and the second displays 1 Days.
[image]
5. Rrepeating steps 1 through 4 will display 1 Minutes and 2 Days.
[image]

I have worked arount this issue by adding an intervalValue N-number field and an intervalUnit S-select field in the interval field's place then calculating the interval in the sketch.

Thanks for having a further look at this when time and priorities permit.

Thanks!

Onno-Dirkzwager, 2021-04-13:
@larz99 please retry...should be fixed now.
Don't forget to clear cache and refresh....

larz99, 2021-04-17:
I've retested and still get incorrect results.

I've included the debug log at level 3.  The log shows the error occuring prior to the interval value is being written to EEPROM.

_Enter config mode and visit App Settings tab:_

05:59:40:584 ->  Serving device info
05:59:44:864 ->  Serving App Settings
05:59:45:525 -> [{"l":"Room", "v":"Office", "n":"0", "m":"32", "t":"L"},{"l":"MQTT Server", "v":"192.168.1.10", "n":"1", "m":"32", "t":"L"},{"l":"MQTT Port", "v":"1883", "n":"2", "m":"7", "t":"N"},{"l":"MQTT User", "v":"Larz", "n":"3", "m":"32", "t":"L"},{"l":"MQTT Password", "v":"LarzMqttPassword", "n":"4", "m":"32", "t":"L"},{"l":"MQTT Topic", "v":"home", "n":"5", "m":"32", "t":"L"},{"l":"Run Mode:Powered, Realtime, Store & Forward", "v":"1", "n":"6", "m":"1", "t":"S"},**{"l":"Update Interval", "v":"36000", "n":"7", "m":"7", "t":"I"},{"l":"Call Home Interval", "v":"43200", "n":"8", "m":"7", "t":"I"}**,{"l":"Timezone", "v":"Europe/London", "n":"9", "m":"48", "t":"Z"}]

_Set interval1 to 1 minute
Set Interval2 to 1 minute
Press Save_

06:01:17:701 ->  Saving App Settings
No need to overwrite current value
No need to overwrite current value
No need to overwrite current value
No need to overwrite current value
No need to overwrite current value
No need to overwrite current value
No need to overwrite current value

Overwrite with new value: 60
EEPROM from: 991 to 999

Overwrite with new value: 60
EEPROM from: 1011 to 1019
No need to overwrite current value

_AOK

Refresh browser
Revisit App Settings tab_

06:02:31:707 ->  Serving device info
06:02:38:380 ->  Serving App Settings
06:02:39:040 -> [{"l":"Room", "v":"Office", "n":"0", "m":"32", "t":"L"},{"l":"MQTT Server", "v":"192.168.1.10", "n":"1", "m":"32", "t":"L"},{"l":"MQTT Port", "v":"1883", "n":"2", "m":"7", "t":"N"},{"l":"MQTT User", "v":"Larz", "n":"3", "m":"32", "t":"L"},{"l":"MQTT Password", "v":"LarzMqttPassword", "n":"4", "m":"32", "t":"L"},{"l":"MQTT Topic", "v":"home", "n":"5", "m":"32", "t":"L"},{"l":"Run Mode:Powered, Realtime, Store & Forward", "v":"1", "n":"6", "m":"1", "t":"S"},**{"l":"Update Interval", "v":"60", "n":"7", "m":"7", "t":"I"},{"l":"Call Home Interval", "v":"60", "n":"8", "m":"7", "t":"I"}**,{"l":"Timezone", "v":"Europe/London", "n":"9", "m":"48", "t":"Z"}]

_AOK

Set Interval2 to 12 hours
Set interval1 to 10 minutes
Press save_

06:04:44:036 ->  Saving App Settings
No need to overwrite current value
No need to overwrite current value
No need to overwrite current value
No need to overwrite current value
No need to overwrite current value
No need to overwrite current value
No need to overwrite current value

Overwrite with new value: 36000
EEPROM from: 991 to 999

Overwrite with new value: 43200
EEPROM from: 1011 to 1019
No need to overwrite current value

_Note Interval1 is incorrect_

_Press OK
Browser display is OK
Refresh browser & revisit App Settings_

06:06:12:823 ->  Serving App Settings
06:06:13:490 -> [{"l":"Room", "v":"Office", "n":"0", "m":"32", "t":"L"},{"l":"MQTT Server", "v":"192.168.1.10", "n":"1", "m":"32", "t":"L"},{"l":"MQTT Port", "v":"1883", "n":"2", "m":"7", "t":"N"},{"l":"MQTT User", "v":"Larz", "n":"3", "m":"32", "t":"L"},{"l":"MQTT Password", "v":"LarzMqttPassword", "n":"4", "m":"32", "t":"L"},{"l":"MQTT Topic", "v":"home", "n":"5", "m":"32", "t":"L"},{"l":"Run Mode:Powered, Realtime, Store & Forward", "v":"1", "n":"6", "m":"1", "t":"S"},**{"l":"Update Interval", "v":"36000", "n":"7", "m":"7", "t":"I"},{"l":"Call Home Interval", "v":"43200", "n":"8", "m":"7", "t":"I"}**,{"l":"Timezone", "v":"Europe/London", "n":"9", "m":"48", "t":"Z"}]

_Note interval1 is 36000

App Settings tab shows interval1 as 10 hours_

Onno-Dirkzwager, 2021-04-17:
Annoying, I tested with tens of values and had no issues saving your previous (screenshot) values.

But you've made your case...it's clear it's not perfect yet. I'll retest with your latest values.