iotappstory/ESP-Library #187: Not work for ESP8266 Arduino core 3.0.0

lenguyenvu007, 2021-06-20:
Let's consider to build a new version work for ESP8266 Arduino core version 3.0.0
Error: ESP-Library\src\espressif\esp8266\callServer_WiFiClientSecure.cpp:175:17: error: 'class BearSSL::WiFiClientSecure' has no member named 'verify'

klasrocket, 2021-08-06:
I got a similar problem and took me a little time to figure out the problem.
setup:
 Platform.io and vs code on ubuntu 21.10 beta
esp8266 nodemcu
running the example file IASLoader.ino gave me the following error;

.pio/libdeps/nodemcuv2/IOTAppStory-ESP/src/espressif/esp8266/CallServer.cpp:182:35: error: 'class BearSSL::WiFiClientSecure' has no member named 'loadCACert'

after downgrading the platform; Espressif 8266 from 3.0.0 to 2.6.3 I got it to compile properly though.
seems like there where quite a few brakeing changes so not suprising; https://github.com/esp8266/Arduino/releases/tag/3.0.0

Mergenth, 2021-11-07:
To compile properly though, it is sufficient downgrading (Arduino Boards Manager) ESP8266 from 3.0.0 to 2.7.4 

Now IDE is properly compiling again.  ;-)