iotappstory/ESP-Library #188: Redefinition of RTCMEMBEGIN in v2.1.0 RC5.1

SiskoBen, 2021-06-26:
In v2.1.0 RC5.1 RTCMEMBEGIN has two conflicting definitions, both https://github.com/iotappstory/ESP-Library/blob/2.1.0-RC5.1/src/espressif/esp8266/BoardInfo.h and https://github.com/iotappstory/ESP-Library/blob/2.1.0-RC5.1/src/espressif/esp8266/NtpTimeSync.h are defining RTCMEMBEGIN.

In the BoardInfo.h RTCMEMBEGIN is defined as 68 while in the NtpSync.h RTCMEMBEGIN is defined as 81.

Looking at the corresponding BoardInfo.cpp and NtpSync.cpp files where the RTCMEMBEGIN is being used/referenced I deduced that each of the RTCMEMBEGIN definitions is actually intended to address a different part of RTC memory but having RTCMEMBEGIN redefined will probably mean that only one RTCMEMBEGIN value is used throughout the code by the compiler and thus each of the two usages will overwrite the others data structure, note that it also does not help that the MAGICBYTE used for each of the structure usages is defined as the same (85) thus each of the two usages of the RTC memory location (which ever the compiler choses to use) will both have the same MAGICBYTE and thus cannot determine whether it is actually referencing the correct data structure or the data structure of the other.
My suggestion is to change the RTCMEMBEGIN in the NtpSync.h and NtpSync.cpp files to a different name (such as NTPRTCMEMBEGIN) and also to have the MAGICBYTE in both NtpSync.h and NtpSync.cpp files be renamed to NTPMAGICBYTE and also to use a different unique NTPMAGICBYTE value.

Onno-Dirkzwager, 2021-07-03:
@SiskoBen I already answered you on Discord but you might have missed that. I agree with your observations and will implement your suggestions in the next release. Thank you for pointing this out!

kobusnwu, 2024-04-26:
@Onno-Dirkzwager I seem to have run into the same problem now, was this ever implemented?